controlpaths/hackbat #1: More info?

RonRN18, 2024-05-12:
I saw a blurb about this project but having difficulty finding any more information about it. From this Github, I see how the board can be built but as of yet, I have not seen any software to use on the hardware. Is more information forthcoming, such as instructions for programming and pointers on what can be accomplished with it?

rdfry, 2024-05-14:
I am also interested in programming this.

controlpaths, 2024-05-14:
Hi, this project is a HW platform. The board can be programmed using the Arduino IDE. All the peripherals except the RF have an Arduino library available so you can use it. For the RF IC, you need to modify the library from LSatan (SMARTRC-CC1101-Driver-Lib), or download the library from the repository of Joel Serna Moreno, who has a fork of the library modified for the RP2040.

Thanks!

closed 2024-05-16 by controlpaths