controlpaths/hackbat #3: Bom file error

ygaspoz, 2024-05-15:
Hi, when uploading the files for the parts JLCPCB gives me the following error: 
Error

The below parts won't be assembled due to data missing.
U5,J1,R18,R17 designators don't exist in the BOM file.
AE1 designator don't exist in the CPL file.

What is the fix?

fabinhojcr, 2024-05-15:
Upload the hackbat.zip file and you will be successful.

ygaspoz, 2024-05-15:
> Upload the hackbat.zip file and you will be successful.

I tried doing that, but when it prompts me to upload that BOM and CPL files I still get the same error.

TenThousand1, 2024-05-15:
I am also getting the exact same error message.

shayfer, 2024-05-15:
Same error message as well. It does make sense though since the hackbat-all-pos.csv references those items and they are not in the bom.csv file and vice versa for the AE1.

ygaspoz, 2024-05-16:
OK so i am correct in assuming that the display, antenna and NFC module need to be bought separately?

controlpaths, 2024-05-16:
That is correct. For the antenna, you can solder the antena or an SMA connector to attach an external antenna.

shayfer, 2024-05-16:
@controlpaths but you can add the antenna, LCD, etc to the bom & centroid file and have them do it correct?

MemomiAlexCherkashin, 2024-05-17:
What is final price of this assembly?
I've got something like:
Charge Details
PCB Price $7.00 
Via Covering: $0.00
Special Offer: $7.00
Standard PCBA Price $163.22 
Setup Fee: $50.00
Stencil: $15.71
Panel: $0.00
Large Size: $0.00
Components(30 items): $37.98
Feeders Loading fee:  $36.46
SMT Assembly $2.86
Hand-soldering labor fee: $3.50
Manual Assembly  $0.38
PCB assembly fixture  $15.71
Packaging fee: $0.47
Special components fee: $0.15
**Total Price:
$170.22**
Which does not include LCD Antenna, case etc.. 
So based on https://www.xda-developers.com/raspberry-pi-hackbot/ : 
"However, unlike the Flipper Zero, the Hackbat is both open-source and cheaper than its cousin's $169 price tag." 
Why???

chaton78, 2024-05-17:
Not sure what is your setup.. .my setup fee is 8$ not 50 and I took economic PCBA... 73$

MemomiAlexCherkashin, 2024-05-17:
> Not sure what is your setup.. .my setup fee is 8$ not 50 and I took economic PCBA... 73$

I managed to achieve this value by skipping of replacement for capacitors (which could not be found in the stock by default):
4,"C13, C21",2,3.9p,Capacitor_SMD:C_0402_1005Metric,,
5,"C14, C17",2,220p,Capacitor_SMD:C_0402_1005Metric,,
6,C15,1,8.2p,Capacitor_SMD:C_0402_1005Metric,,
7,C16,1,5.6p,Capacitor_SMD:C_0402_1005Metric,,
8,C18,1,100n,Capacitor_SMD:C_0402_1005Metric,,
And use economic PCBA like you say, but it means that this capacitors and TF port / RF module need to be soldered later alongside with antenna and display

jradm87, 2024-05-20:
Hello, i have problem too on the boom file:

Error

The below parts won't be assembled due to data missing.
U5,J1,R18,R17 designators don't exist in the BOM file.
AE1 designator don't exist in the CPL file.

[image]

Zeddex, 2024-05-21:
I got the same error:
The below parts won't be assembled due to data missing.
U5,J1,R18,R17 designators don't exist in the BOM file.
AE1 designator don't exist in the CPL file.

Is there fixed files?

MeanGreen420, 2024-05-23:
Ill try to help some,

When you get the Following error

**The below parts won't be assembled due to data missing.
U5,J1,R18,R17 designators don't exist in the BOM file.
AE1 designator don't exist in the CPL file.**

its because components are not intended to be installed by JLCPCB, you can install them yourself
U5 is the Display (DM-OLED096-636) as stated above
J1 is the 4 pins that connect the display that usually come with it if you order it from ebay or something.

R18 & R19 are not populated because they are used to change polarity for the display as explained in the documentation
[image]

AE1 is the antenna location, can be populated with a sma connector or a simple internal antenna, I haven't found a source for them yet.

I did also have to substitute many capacitors on my order because of supply shortages probably due to this project. I was able to find compatible substitutes for all of them though, this may be challenging if your not familiar with electronic components.

Hope this helps!

closed 2024-05-29 by ygaspoz