controlpaths/hackbat #4: RFID/IR support?

MonoTheProtogen, 2024-05-15:
I've basically heard about this today and immediately found this, and am really interested in this, now I noticed that the NFC module used in this project also supports RFID, would this also be the case with the device itself or would a separate integration need to be written into the firmware? I also wanted to ask, since I've looked through the schematics, would it be possible to use 2 of the 6 unoccupied pins on the RP2040 to add an infrared diode and infrared photoresistor to add infrared capabilities?

An additional idea of mine would be to route some power output pins, ground pins and the remaining 4 GPIO pins outside the board for a sort of expansion header for some stuff the end user might want to make, would these ideas be possible in theory?

controlpaths, 2024-05-16:
Hi, there is no official FW for the Hackbat, it is a HW platform for experiment with your own FW using the Arduino IDE or the RP2040 IDE so, if the NFC board supports the protocol you need you can use it without problem.

Regarding the IR, yes, you can, but both UART ports are already in use for the ESP programming and the external USB-UART converter. Maybe the best would be eliminate the external USB-UART connector and use it for the IR.

And of course, you can add an external port with the unused pins.

Thanks!

MonoTheProtogen, 2024-05-16:
noted, I'll see what I myself can do about adding IR to the board, having all of the schematics provided, plus the KiCAD files will help, thanks!

closed 2024-05-16 by controlpaths