controlpaths/hackbat #7: please add more keys

zapewne, 2024-05-17:
**Please add https://reticulum.network/ directly. Only this functionality is good idea to but this device. (or meshtastic network)**

- Please add more keys, 16 is ok similar esphone or notkia 
- I need socket for antenas. Sometimes external antena is better way to use than embeded
- This device must have a internal power. Meybe ad stepup down from 1V to 5V? Any AA, AAA, 18650 etc will work.

fachinformatiker, 2024-05-20:
You can find the schematics, PCB, etc in this Repo.
Download it and add the changes yourself, then upload the results :)
Well thank you

controlpaths, 2024-05-22:
Correct, the idea is that anybody could add their improvements and share it. In any case, more keys? Seems that you want a phone, that is not the goal of this project. Anyway, you can add all the keys you need and share it. Thanks!

closed 2024-05-22 by controlpaths

fab1an2, 2024-05-22:
It seems that if you want to enter some data, number and ip mask, or password then you will appreciate more keys.

In my opinion, it is very necessary right away in the first model.

And if you want to write a message? Who knows what people will use it for?