controlpaths/hackbat #10: About library

Yadavbhargav64, 2024-05-22:
Which are library and code should install please give us detail instructions

rdfry, 2024-05-22:
He is only providing this as a hardware project. It is up to the makers to code it

Zaidx1, 2024-05-23:
So there is no code available

rdfry, 2024-05-23:
No

MeanGreen420, 2024-05-23:
if you look in the Closed Issues section here [https://github.com/controlpaths/hackbat/issues/1](url) you will find this

Hi, this project is a HW platform. The board can be programmed using the Arduino IDE. All the peripherals except the RF have an Arduino library available so you can use it. For the RF IC, you need to modify the library from LSatan (SMARTRC-CC1101-Driver-Lib), or download the library from the repository of Joel Serna Moreno, who has a fork of the library modified for the RP2040.

closed 2024-05-27 by controlpaths