controlpaths/hackbat #12: Can i make it with pi pico w board

Zaidx1, 2024-05-23:
Is it possible to hook up the modules like RFID sensor CC1101 and oled display with raspberry piko w board and connect ESP 32 as well with wires and program both board separately by using their USB port or we can only use PCB version of this flipper zero

controlpaths, 2024-05-27:
Hi, yes, you can buy the boards separately and it will work. This project just build in one board different development boards. You can find more info on the hackster project.

Zaidx1, 2024-06-02:
Ok. But where is the code file

controlpaths, 2024-07-24:
Tou can use Arduino libraries for all the components of the board. I do not provide the code because I just tested all the peripherals but I don't have an application code

closed 2024-07-24 by controlpaths