controlpaths/hackbat #15: Display Size

MeanGreen420, 2024-06-01:
I just got my batch of boards and noticed the .96" Display doesn't fit the outline and pins, it looks like a 1.3" would fit better.  Anyone else notice this?

bfpa40, 2024-06-01:
Im glad I just read this as I am almost prepared to order from JLCPCB. If Im reading the main post correctly the schematics reflect the DM-OLED096-636 for the .96" Display then pin 1 = Ground, Pin 2 = VCC, if you choose to use the SSD1306 then pin 1 = Ground and pin 2 = VCC. Correct?

rdfry, 2024-06-01:
yeah it would be nice to know what the overall dimentions of the lcd bd is. the profile is 36mm x 34mm

bfpa40, 2024-06-01:
Ive now looked at both displays and they both say .96" must be the pin layout is different i assume?

rdfry, 2024-06-01:
except the screen shown is almost the size of the bd profile which is 1.5 " wide x  1.325 " tall

bfpa40, 2024-06-01:
Here is what I am showing:
For the SSD1306
(26.70mm x 19.26mm x 1.85mm) or (1.051181”  x  0.7582677”  x  0.07283465”)

For the DM-OLED096-636
(26mm x 26mm x 2.7mm) or  (1.023622”  x  1.023622”  x 0.1062992”)

rdfry, 2024-06-01:
that just dosent seem to be big enough. i posted a question in the ada fruit forum. i love ada fruit but im not paying what they want for a screen. Ali Express if i can find the screen that fits

bfpa40, 2024-06-01:
I understand a little better now you want a larger screen... Ive got my order in the cart on JLCPCB but im at a halt now because there isnt a baseline firmware/software image to flash to it as of yet and I havent had the greatest support with folks online trying to get info to get new devices up and running..so Im at a standstill

rdfry, 2024-06-01:
i know i'm just going off the picture of the bd  that has the screen on it. i have the bds from jlcpcb. and the bd with the screen has to be 1.5" X 1.325" overall.  i have a .96 Screen and the bd is way to small. it is just frustrating that the vendors dont provide the bd dimentions  along with the screen size

MeanGreen420, 2024-06-01:
my .96" displays are about 1 1/16th  so im guessing the 1.3 display will be close to 1.5, also all the 1.5" displays I have seen have the connector on the side not the top.

rdfry, 2024-06-01:
my .96" display bd measures 1" square

MeanGreen420, 2024-06-02:
the outline of the display on the board only measures 1 3/8" also so its not 1 1/2"

rdfry, 2024-06-02:
your right i cant read a ruler. 1 3/8 x 1 5/16

rdfry, 2024-06-08:
Received a 1.3 display the mounting holes are slightly off but the board fits the outline

MeanGreen420, 2024-06-09:
same here, just got mine today. just to be clear since I didn't state it very clear, the 1" display connection pins do match up, its the mounting holes and outline that didn't match, the 1.3" display is much better, I was able to use m2.5 screws and nuts to secure mine to the board also.

bfpa40, 2024-06-09:
So which display is the 1.3" display do you have a P/N  to insure i get the correct one?

MeanGreen420, 2024-06-09:
Product Introduction:
1.Size:1.3 inch
2.Resolution:128*64
3.Controlling Chip:SSH1106
4.Display Area:29.42*14.7mm
5.Driving Voltage:3.3-5V
6.Operating Temperature:-40~70 celsius
7.Interface Type:IIC
8.Light Color: white

[https://www.ebay.com/itm/186114101614](url)

rdfry, 2024-06-09:
Just found this amazing item on AliExpress. Check it out! 
$2.87 | 1.3 Inch 1.3" White OLED LCD 4PIN Display Module IIC I2C Interface 128x64 for Arduino
https://a.aliexpress.com/_msY2vIm

bfpa40, 2024-06-09:
If anyone in here has ordered there boards where did you order them from? I have 5 in the cart now at JLCPCB total cost for all 5 making and populating these boards there is $176.77  thats  $35.34 each

rdfry, 2024-06-09:
I just had jlcpcp make the boards. Cost about 15 dollars. I am doing my own smt. You really only need the components on one of the boards unless you want 5 complete boards.

bfpa40, 2024-06-09:
I want 5 complete boards so I can have 2 and sell the other 3 to recoup some $$

MeanGreen420, 2024-06-09:
I had them only populate the front side of the board, its much more expensive to have them do both, the esp12 is pretty easy to solder on, the sd card holder can be done with a iron if your careful, I used my hot air rework station.  Also if your trying to recoup some money have them make 10, an additional 5 board is a fraction of the cost, I only had to pay like $25 more dollars to make 5 more.

bfpa40, 2024-06-15:
I Received my OLED displays in the 1.3" and I tried them using a RaspberryPi Pico and while I thought they were of the ssd1306 flavor driver I couldnt get them working until i used the sh1106 version driver in the pico. Just wanted to add this here in case anyone else was having issues

closed 2024-07-24 by controlpaths