controlpaths/hackbat #17: BOM / CPL Error @ JLCPCB

boopboopboopers, 2024-06-03:
The following are given when uploading BOM and CPL files.

"Error

The below parts won't be assembled due to data missing.
U5,J1,R18,R17 designators don't exist in the BOM file.
AE1 designator don't exist in the CPL file."

SanchosPoncho, 2024-06-06:
Please refer to https://github.com/controlpaths/hackbat/issues/3

These are materials that are provide by installing the screen, NFC, and radio antenna to the board that have to completed after the fact.

boopboopboopers, 2024-06-07:
Well, I feel like a dufus. Makes total sense. Got ahead of myself. Thanks for the response.

bfpa40, 2024-06-15:
Whats the voltage value for Capacitor C18? Anyone know?

controlpaths, 2024-07-24:
6.3V is enough for any capacitor.

closed 2024-07-24 by controlpaths