controlpaths/hackbat #21: JLCPCB Has an issue with C18 not fitting footprint

bfpa40, 2024-06-13:
Here is what they sent me. What did other folks use successfully for C18?

[image]

bfpa40, 2024-06-13:
Evidently no one else had issues with this component not fitting the footprint using JLCPCB and if you did what did you substitute it with? I still need assistance with this please someone HELP !?!?!

MeanGreen420, 2024-06-21:
I used C424821

Fdelro, 2024-08-12:
I have a similar issue, when I upload the Geber files and in the next step the bom.csv and the hackbat-all-pos.csv I get an error message with the following: Error
The below parts won't be assembled due to data missing.
U5,J1,18,R17 designators don't exist in the BOM file.
AE1 designator don't exist in the CPL file.