controlpaths/hackbat #22: HELP Needed

bfpa40, 2024-06-14:
JLCPCB after ordering emailed me back and said the footprint of C18 wont fit the pads it looks to be way small in the picture they sent me. What is the value of that capicitor? I know its 100 nano farads but how many volts should it be rated  for??  Better yet does anyone have a reference on the JLCPCB Part# they ordered that worked? My order is on hold until i get this part corrected.
The Part they have which came from the BOM says its as follows:

6.3V 100nF X5R ±10% 01005 Multilayer Ceramic Capacitors MLCC - SMD/SMT ROHS

I see the problem is that the package is 01005 instead of 0402 but before i correct it I want to insure the values of the capacitor are correct (110n , 6.3V)  is that right Yes/No    and Thank You

controlpaths, 2024-07-24:
6.3 volts is enough for any capacitor. The board works at 3v3.

gh05t5, 2024-11-24:
Hello! I've uploaded the BOM and centroid files but when they get processed I get this error message: 

**The below parts won't be assembled due to data missing.
U5,J1,R18,R17 designators don't exist in the BOM file.
AE1 designator don't exist in the CPL file.**

Please advise!

Thank you very much

WhaleWithCake, 2024-11-24:
> Hello! I've uploaded the BOM and centroid files but when they get processed I get this error message:
> 
> **The below parts won't be assembled due to data missing. U5,J1,R18,R17 designators don't exist in the BOM file. AE1 designator don't exist in the CPL file.**
> 
> Please advise!
> 
> Thank you very much

See the issue resolved here: https://github.com/controlpaths/hackbat/issues/3