controlpaths/hackbat #26: Firmware  ?

Brunoido90, 2024-07-25:
Hello, I can't not find the firmware?

rdfry, 2024-07-26:
There isn't any. This is provided as a hardware project only. It's up to the end user to make your own firmware.

dan91ng, 2024-08-13:
I have completed the soldering. Can you provide a test firmware to let me know that my soldering is OK?

Fdelro, 2024-08-23:
Does anyone know if the flipper zero firmware can be used on the Hackbat?

rdfry, 2024-09-05:
> Does anyone know if the flipper zero firmware can be used on the Hackbat?

No this uses a rpi 2040 MCU

Variable-61, 2025-03-07:
> There isn't any. This is provided as a hardware project only. It's up to the end user to make your own firmware.

Seriously! You designed the hardware without any firmware? How did you test your hardware, or did you just simulated it! This is ridiculous.